proveskit/flight_controller_board #22: Clock speed changes during runtime

TheSilentDefender, 2024-10-04:
Pico SDK enables dynamic adjustment of the clock speed at runtime. With some modifications to the board-specific files provided in CircuitPython, this functionality can be exposed on the Python side of the firmware.

This could potentially reduce power consumption when the clock speed is lowered during non-intensive periods of orbit.

TheSilentDefender, 2024-10-04:
Branch flight_controller_board should have the updated board files for circuit python, and a new firmware file.

Mikefly123, 2024-10-11:
# Power Consumption Test Procedure
1. Attach the RP2040 to a power supply
2. Measure the power consumption at various points

To change the clock frequency of the microcontroller use the following code block: 
```py 
import machine, microcontroller
microcontroller.cpu.frequency
machine.set_clock(INSERT CLOCK SPEED IN HZ)
microcontroller.cpu.frequency
```
We need to collect data at the following points: 
| Clock Speed (MHz) | Sample 1 (mW) | Sample 2 (mW) | Sample 3 (mW) | Sample 4 (mW) |
|-------------------|---------------|---------------|---------------|---------------|
| 125               |               |               |               |               |
| 100               |               |               |               |               |
| 75                |               |               |               |               |
| 50                |               |               |               |               |
| 25                |               |               |               |               |

wrong-formatt, 2024-10-12:
Power supply used: Korad KA3005P Programmable DC POwer Supply. 
Units of precision: Amps to 3 decimal places
Datatable will be updated

| Clock Speed (MHz) | Sample 1 (mW) | Sample 2 (mW) | Sample 3 (mW) | Sample 4 (mW) |
|-------------------|---------------|---------------|---------------|---------------|
| 125               |        151.8       |     125.4          |               |               |
| 100               |       151.8        |     125.4        |               |               |
| 75                |       151.8        |      125.4        |               |               |
| 62.5             |      118.8         |      89.1          |               |               |
| 50                |       115.5        |      85.8        |               |               |
| 25                |       102.3        |       72.6        |               |               |
| 12                |       99             |      69.3         |               |               |
| 1                  |       89.1          |       62.7        |               |               |

Mikefly123, 2024-10-12:
Graphed by Claude: 
[image]

Mikefly123, 2024-10-15:
Closing this for now and continuing to track clock speed stuff in the linked flight software repo!

commit 442165b2 (2024-10-08): fixed #22

hopefully improves power consumption
closed 2024-10-15 by Mikefly123