proveskit/flight_controller_board #24: InspireFly Special Watchdog Interval is Unexpectedly Fast

Mikefly123, 2024-10-19:
# Description of the Issue
When biased with a 5V inout from a bench top PSU the watchdog appears to reset every 5 seconds when not being pet rather than the expected 26s. We are investigating this further.

Mikefly123, 2024-10-22:
Just spoke with Andrew Greenberg and the apparent culprit it BOM consolidation! 
[image]
Specifically, the D7 diode in the watchdog circuit was consolidated to be a common Schottky Diode in the InspireFly special and the voltage leakage through it is what is most likely changing the period of the oscillator. A note has been added to the schematic that we CANNOT consolidate the BOM on this sheet.

polygnomial, 2024-10-23:
👀

Mikefly123, 2024-10-31:
Closing this issue for now, it will be fixed on V5 Flight Controller Boards!

closed 2024-10-31 by Mikefly123