proveskit/flight_controller_board #26: VirginiaTech V1.2 Bus Burnout Issue?

Mikefly123, 2025-01-23:
# Description of the Problem 
The Virginia Tech team is reporting that on their RP2040 boards they are experiencing premature failures of their RP2040's due to what is believed to be an overvolted `1.2V` bus which floats at `1.8V` instead. Apparently, the failure rate is as high as 90%!!!

lkv0315, 2025-01-29:
This is effectively resolved, issue was due to the RP2040's ground pad being isolated from the main ground net due to a pin naming discrepancy between our schematic symbol and board footprint. We have not been able to test a board with this pad grounded but we will be ordering one shortly.

Mikefly123, 2025-01-30:
Amazing! Thanks for the update!

closed 2025-01-30 by Mikefly123