proveskit/flight_controller_board #29: Merging firmware builds into circuitpython upstream

nateinaction, 2025-03-09:
Hey team,

I'm working on merging the CircuitPython firmware builds into upstream. I have been able to successfully build the v1 firmware (without the system clock changes) but I'm having some difficulty building the v2 firmware with the clock changes.

https://github.com/adafruit/circuitpython/commit/f4c0176b260a554a7f11cd04815adf19e1411491
```
make BOARD=proveskit_rp2040_v4
- Verbosity options: any combination of "steps commands rules", as `make V=...` or env var BUILD_VERBOSE
GEN build-proveskit_rp2040_v4/genhdr/mpversion.h
QSTR updated
GEN build-proveskit_rp2040_v4/genhdr/qstrdefs.generated.h
Module registrations updated
GEN build-proveskit_rp2040_v4/genhdr/moduledefs.h
Root pointer registrations updated
GEN build-proveskit_rp2040_v4/genhdr/root_pointers.h
   text	   data	    bss	    dec	    hex	filename
    244	      0	      0	    244	     f4	build-proveskit_rp2040_v4/boot2.elf
Memory region         Used Size  Region Size  %age Used
  FLASH_FIRMWARE:      857896 B      1020 KB     82.14%
             RAM:       63208 B       256 KB     24.11%
       SCRATCH_Y:          0 GB         4 KB      0.00%
       SCRATCH_X:          2 KB         4 KB     50.00%
Converted to uf2, output size: 1716224, start address: 0x10000000
Wrote 1716224 bytes to build-proveskit_rp2040_v4/firmware.uf2
```

https://github.com/adafruit/circuitpython/compare/main...proveskit:circuitpython:proveskit-rp2040-v4-2?expand=1
```
make BOARD=proveskit_rp2040_v4
...
In file included from boards/proveskit_rp2040_v4/pins.c:9:
boards/proveskit_rp2040_v4/clock.c:20:24: error: 'machine_module' defined but not used [-Werror=unused-variable]
   20 | static mp_obj_module_t machine_module = {
      |                        ^~~~~~~~~~~~~~
cc1: all warnings being treated as errors
-e See https://learn.adafruit.com/building-circuitpython; Adafruit Discord #circuitpython-dev
make: *** [build-proveskit_rp2040_v4/boards/proveskit_rp2040_v4/pins.o] Error 1
```

These issues led me down the road of trying to see if any other CircuitPython RP2040 boards have this functionality, they don't seem to. But why?

If I'm reading the documentation correctly it seems that we can adjust CPU clock speed directly in CircuitPython without creating our own `machine` module. Instead we can set the frequency using [`microcontroller.Processor.frequency = some value`](https://docs.circuitpython.org/en/latest/shared-bindings/microcontroller/index.html#microcontroller.Processor.frequency). Have we tried that?

FYI @Mikefly123

Mikefly123, 2025-03-10:
Paging @TheSilentDefender in to help with this since he originally developed the clock speed firmware! 

@nateinaction I actually did just test a board right now and you can indeed just use `microcontroller.cpu.frequency = SOME VALUE` to set the frequency directly! This would mean we don't need to use the machine module in the future.

An interesting thing we just realized while testing is you can actually address each of the two cores individually and change their clockspeed. I don't have a power supply here to check if individually reducing the clock speed has a big effect on power consumption, but maybe it does! 
[image]

Mikefly123, 2025-03-27:
Closing this out, as we have now indeed merged with upstream CircuitPython!!!

closed 2025-03-27 by Mikefly123